ChristopheJacquet/PiFmRds #145: piping pulseaudio

augisbud, 2020-04-22:
Hello, I have been experimenting and trying to pipe bluetooth audio input into pifmrds, here is the command I'm trying to use
`sudo -u pi pacat -r -d 0 --latency-msec=50 | sudo -u pi sox -t raw -r 44100 -e signed-integer -b 15 -c 2 - -t wav - | sudo ./pi_fm_rds -audio -`
however I this is the output
`Using mbox device /dev/vcio.
Allocating physical memory: size = 3403776     mem_ref = 14     bus_addr = 5e78f000     virt_addr = 0xb68c4000
sox WARN wav: Length in output .wav header will be wrong since can't seek to fix it
sox FAIL sox: `-' this handler does not support this data size: Unsupported data format
ppm corr is 0.0000, divider is 1096.4912 (1096 + 2012*2^-12) [nominal 1096.4912].
Using stdin for audio input.
Input: 44100 Hz, upsampling factor: 5.17
2 channels, generating stereo multiplex.
Created low-pass FIR filter for audio channels, with cutoff at 12000.0 Hz
PI: 1234, PS: <Varying>.
RT: "PiFmRds: live FM-RDS transmission from the RaspberryPi"
Starting to transmit on 107.9 MHz.
Could not rewind in audio file, terminating
Terminating: cleanly deactivated the DMA engine and killed the carrier.
write() failed: Broken pipe
`

mirkokral, 2022-12-03:
Sounds like a sox issue.

KubaPro010, 2023-11-11:
Instaad of pacat and sox, use parecord, so you'd type: `parecord -d 0 --latency-msec=50 --raw | sudo ./pi_fm_rds -audio -`