ChristopheJacquet/PiFmRds #146: Possible to go down to 45Mhz?

danielmoore123, 2020-04-27:
I am wanting to test RDS pass-through on a dummy load at 48.3MhZ for a studio link, this is the only Pi project I can find that can generate a decent RDS signal, are you able to assist how this program can be forced to go that low?
Kind Regards

tst403, 2020-11-14:
Yes just do the following in **src/pi_fm_rds.c**

Replace this:

            if(carrier_freq < 76e6 || carrier_freq > 108e6)
            fatal("Incorrect frequency specification. Must be in megahertz, of the form 107.9, between 76 and 108.\n");

With those lines:

            if(carrier_freq < 1e6 || carrier_freq > 108e6)
            fatal("Incorrect frequency specification. Must be in megahertz, of the form 107.9, between 1 and 108.\n");

ChristopheJacquet, 2021-01-04:
I assume the above comment answered the question, so closing.

closed 2021-01-04 by ChristopheJacquet