ChristopheJacquet/PiFmRds #147: Rpi 4

Marrley, 2020-06-24:
### Hello!

I've tried/researched many different types/variants of Pi_FM or RPiTx, but they all seem to yield the same results/problems when it comes to the Raspberry Pi 4.

**[Original Pi_FM](https://github.com/rm-hull/pifm):** No errors but no TX Output

**[PiFmAdv](https://github.com/miegl/PiFmAdv/):** No errors but no TX Output (same as Pi_FM)

**[PiFmRds](https://github.com/ChristopheJacquet/PiFmRds/):** No errors but no TX Output (same as Pi_FM)

**[PiFmRds (modded)](https://github.com/markondej/PiFmRds/):** Very low volume output and lots of buzzing/cracking noises

**[RPiTx](https://github.com/F5OEO/rpitx/):** Not working (as of reports from the [/issues](https://github.com/F5OEO/rpitx/issues/))

I can fairly say I have almost searched the entire internet forum system...

**_Two Questions:_**

_**1:** Does anyone know of/have a modified version of PiFmAdv or RPiTx that will work on the Raspberry Pi 4?_

_**2:** Is it possible to use the Raspberry Pi 4 as a standalone RDS transmitter alongside and external FM Transmitter that does not support RDS?.. or will the frequencies just cancel each other out?_

Thx! 
-Marley

PS: I C/P this from [https://github.com/miegl/PiFmAdv/issues/76](https://github.com/miegl/PiFmAdv/issues/76)

Marrley, 2020-06-26:
Anyone?

ChristopheJacquet, 2020-07-08:
Question 1- I don't know more than you do about the RaspberryPi 4. I guess I need to get one myself to experiment :) I'll de-duplicate this issue against the one opened for this issue.

Question 2- No, it won't work. The audio and the RDS data must be part of the same FM modulation.

ChristopheJacquet, 2020-07-08:
Duplicate of 134.

ChristopheJacquet, 2020-07-08:
Duplicate of 134

ChristopheJacquet, 2020-07-08:
Duplicate of #134

ChristopheJacquet, 2021-01-04:
Any followup will happen on #134.

closed 2021-01-04 by ChristopheJacquet