ChristopheJacquet/PiFmRds #150: Why only 76 to 108 Mhz range?  That makes no sense?

dachshund-digital, 2020-07-12:
Why only 76 to 108 Mhz range?  That makes no sense?  There is so much more that this solution could be used for beyond that range.

gshingles, 2020-07-12:
It makes a lot of sense. Most countries have strict regulations on which frequencies you can cause equipment to transmit on.
I think if you check the readme for most, if not all, of the Pi FM transmitter projects out there you'll find a rider that you have to use the code legally. 

I'm my country (NZ), this is the list of general user frequencies and the conditions they must be used under:
https://gazette.govt.nz/notice/id/2019-go1588
Note, very few of them are used with FM modulation for broadcasting audio! If you arbitrarily go outside the bands set aside in your locality you run the risk of causing some techs a massive headache, possibly interrupting critical communications, as well as opening yourself up to prosecution. Be very very careful :) 

There's probably also technical reasons, like what the DMA controller is able to clock out etc. and others might be able to answer that on a theoretical basis. I'm pretty sure there will be no appetite for changing this code to do anything other than audio, low power on the FM broadcast band though.

ChristopheJacquet, 2020-07-12:
There are other transmitter programs for the Raspberry Pi that can use other frequency ranges and other modulation modes. You could check out https://github.com/F5OEO/rpitx for instance -- it includes PiFmRds btw.

The whole point of PiFmRds is to produce a Wideband FM + RDS modulation, and I don't think there are FM + RDS receivers outside the 76-108 MHz range. That being said, if you want to extend the range, just change the frequency check at https://github.com/ChristopheJacquet/PiFmRds/blob/master/src/pi_fm_rds.c#L570

closed 2020-07-12 by ChristopheJacquet

dachshund-digital, 2020-07-14:
Frankly, I disagree, that it is your responsibility to police signal ranges.  So you create a solution that is crippled for everyone else?  If someone uses this software incorrectly, that should not be your concern.  IMHO.  This is does not mean I don't understand your point, I just disagree with it.

Where I live there are little to no restrictions and on private property zero restrictions.  As long as I limit the signal power, thus absolute range, to my private property, I don't have any issues at all.  Well, other than the fact that I can't use your solution.

This is like a gun manufacturer designing a weapon that will only fire at the ground when no one is within 10 feet of the weapon.   Because someone might actually shoot someone some day.  

As for RPiTx, I have been using it, but it is no longer actively supported, and fails to work consistently on Debian 10, i.e. buster, on various Pi models.  I was looking for something that is actually maintained actively, which you do, which is great, but I can't use your nice solution, given its policed limitations.