ChristopheJacquet/PiFmRds #154: Problem with FIFO

ParadoxalGamer, 2020-08-16:
When I try to interact with the PiFm process with FIFO, nothing seems to work while I followed the ReadMe... ^^

ChristopheJacquet, 2021-01-04:
Is it still blocking you?

wmcgrathled, 2022-05-13:
The FIFO needs to not be closed after the PiFMRDS program is running.  Once closed it does not get opened again.  I am working on a branch and pull request that addresses this and have one running that you can send data to the named pipe anytime during PiFMRDS operation.  It works!

ChristopheJacquet, 2024-02-24:
Pi-FM-RDS was opening the FIFO in non-blocking mode, but with the latest Raspbian, the behavior was: the FIFO had to be opened by the writer before Pi-FM-RDS was started, and not closed after, as pointed out by wmcgrathled. I'm not sure if it was always like this, or if the semantics of opening a FIFO in non-blocking mode changed.

Anyway, it's probably less confusing to open the FIFO in blocking mode, so at Pi-FM-RDS startup, the user is prompted to start the writer process if they haven't done so yet. Expect a commit with this change, and corresponding README update in a few moments.

commit 495757ff (2024-02-24): Open control pipe in blocking mode.

Up to now, Pi-FM-RDS was opening the FIFO in non-blocking mode, but with the latest Raspbian, the behavior was: the FIFO had to be opened by the writer before Pi-FM-RDS was started, and not closed after, as pointed out by wmcgrathled. I'm not sure if it was always like this, or if the semantics of opening a FIFO in non-blocking mode changed.

Anyway, it's probably less confusing to open the FIFO in blocking mode, so at Pi-FM-RDS startup, the user is prompted to start the writer process if they haven't done so yet. This commit is changing this.

Fixes #154
closed 2024-02-24 by ChristopheJacquet