ChristopheJacquet/PiFmRds #156: Audio not Transmitting

RaphaelDark, 2020-08-31:
im using a pi 4 for doing this and im testing this on my phones fm radio (offline) 
Everything else works perfect but the audio samples transmitted by the pi is not coming up on my radio, iam using the sample .wav files witch comes with the software,  

i enter this :  sudo ./pi_fm_rds -freq 92.7 -audio sound_22050.wav

and my output is

Using mbox device /dev/vcio.
Allocating physical memory: size = 3403776     mem_ref = 17     bus_addr = fe2f5000     virt_addr = 0xb6882000
ppm corr is 0.0000, divider is 1644.7368 (1644 + 3018*2^-12) [nominal 1096.4912].
Using audio file: sound_22050.wav
Input: 22050 Hz, upsampling factor: 10.34
1 channel, monophonic operation.
Created low-pass FIR filter for audio channels, with cutoff at 8820.0 Hz
PI: 1234, PS: <Varying>.
RT: "PiFmRds: live FM-RDS transmission from the RaspberryPi"
Starting to transmit on 92.7 MHz.

ChristopheJacquet, 2021-01-04:
There are reports of the program not working on the Pi 4. See #134. Unfortunately I didn't have the opportunity to test myself yet.

ChristopheJacquet, 2021-01-04:
Duplicate of #134

closed 2021-01-04 by ChristopheJacquet