ChristopheJacquet/PiFmRds #159: Baseband frequency is half with raspberry pi 3 model B

evertverduin, 2020-12-30:
I was playing with this software and experienced:
- No stereo
- No RDS
- Audio too slow.

I found out that the complete baseband is at half of it's frequency. The pilot runs at 9.5 KHz, DSB at 19khz, RDS also 2 times lower.

I think there is a mistake with the baseband upsampler or the NCO constants are at the wrong frequency, but this would not explain why the audio is at half it's frequency either.

BR

Evert

andimik, 2021-01-02:
https://github.com/miegl/PiFmAdv/issues/81

I have the same problem.

Maybe an update of Raspbian or its packages caused that?

evertverduin, 2021-01-02:
There is no support, at least, no one of the devs responds. 
Edited 02-01-2021:

My solution was to erase raspberry os! 
install Raspbian Jesse lite, and install gpu_freq=250 in config.txt

DON'T USE MODERN RASPBERRY PI-OS!

Symptoms are:
Baseband signal is at half of it's sampling rate. Audio to slow, pilot on 9.5 khz and rds at rds/2.

Solution -> install Raspbian Jesse. with gpu_freq=250 in config.txt

And IT Will work!

BR

Evert

ChristopheJacquet, 2021-01-04:
I just tried this with the latest Raspberry Pi OS. I get the same symptoms as @evertverduin. Somehow the clocks must be configured (with newer kernels, or with newer OS releases) to run slower than expected. I thought PLLD always runs at 500 MHz, is this wrong?

Anyway, the consequence of this is that with the wrong assumptions, the divider computed for the PWM clock is twice too high here: https://github.com/ChristopheJacquet/PiFmRds/blob/b0f253232b2c5084c51dc81200e6d1a116221ff2/src/pi_fm_rds.c#L412

An easy workaround is to run Pi-FM-RDS with the argument `-ppm 1e6` : a correction of one million parts per million will halve the divider...

So run for example:

```
sudo ./pi_fm_rds -freq 106 -audio sound.wav -ppm 1e6
```

While this mitigates the issue, I will now try to figure out what clock speed changed, and how to go about it.

CainLynn, 2021-01-16:
I have this problem after updating the installation package

evertverduin, 2021-01-16:
Revert to raspbian

ChristopheJacquet, 2021-01-16:
Investigations so far, more is needed:

On my Pi3, according to `/sys/kernel/debug/clk/clk_summary`, `plld_per` is at 500 MHz, as expected, but `pwm` does not inherit from it, which suggests a bad clock setup somewhere.

Also interesting is `vcgencmd measure_clock pwm` which seems to return 4*228 kHz under correct Pi-FM-RDS operation.

stOneskull, 2021-04-01:
so go back to raspian?

SchattenWolf2008, 2021-04-10:
well, a beep between 7KHz and 7,2KHz

crispino1976, 2022-10-25:
inanzitutto ringrazio i[ChristopheJacquet] per aver condiviso il suo programma su github.
Buonasera a voi tutti ho lavorato molto in questo progetto fantastico per fare in modo che possa funzionare anche con  le ultime release di Rasperry pi 4b.
Sono riuscito a farlo funzionare cambiando PLLFREQ in 648 Mhz tirato fuori da pll_per in cat /.../clk_summary
ho inoltre grazie a questo software ho fatto molti esperimenti scoprendo ed analizzando come funziona il 20(log) dbms
Ho fatto delle modifiche e realizzato un filtro PASSA-BANDA per far cadere il segnale con bandwidth di 200 KHz.
Inoltre ho sperimentato e collaudato ogni sincronismo s_pwm1_sta->FULL1+CPU  o con il magico DMA blocks.
Con alcune adeguate accortezze e frequenze il PLLA SQUARE non genera HARMONICS molto invadenti con 16ma.
NON PARTECIPO ATTIVAMENTE A GITHUB PERCHE' NON SO SCRIVERE BENE INGLESE.