ChristopheJacquet/PiFmRds #160: (question) Why is PiFmRds stuck in 76 to 109 MHz?

Ben1701A, 2021-01-20:
And how come when I disable that limiter, the transmission blanks out below 76 or above 109 MHz?

bastianorts, 2022-04-19:
pifmrds is stuck in this range because this is the range that pretty much every fm receiver use i think thats why there should be no need to go above or below

Ben1701A, 2022-04-28:
> pifmrds is stuck in this range because this is the range that pretty much every fm receiver use i think thats why there should be no need to go above or below

True...I was wondering about getting past that limit since rpitx for the pi4 isn't stable yet (GPU weirdness when I try anything other than `tune`).
I got an alternative past this, though, so I think I'm good here. Thanks for replying. o>o

closed 2022-04-28 by Ben1701A