ChristopheJacquet/PiFmRds #162: how long the audio will repeat

zhouhaohua, 2021-02-08:
any parameter to enable the  loop number, I have tried 
sudo ./pi_fm_rds -audio sound.wav

it run forever, actually I want control the repeat time , how to do that?

Lucstay11, 2023-02-05:
Hi I was in the same situation as you I wanted to play the sound only once because by default it is played in an infinite loop. So I inspected the code and I understood the problem.
At line **487** of **the src/pi_fm_rds.c** file, it makes an infinite loop that goes through the song file like this:` for (;;) { code...}`
So I tested several techniques to stop the sound file when it is finished and I found to loop up to the size in KO of the file so replace the ` (;;)`  by `(int i=1; i<ppm; i++)`.
The **ppm** variable is an argument variable that we hardly ever use so I took it.
So once that change you have to recompile all that in the src folder:
```
$ src/
sudo make clean
sudo make
```
So now use that like this:
```
size_file=$(du song.wav)// Store sound file size in a variable
sudo ./pi_fm_rds -freq 105 -ps "Hacked!" -rt "Hello word" -ppm $size_file -audio song.wav
```
And hop the sound stops at the last second.
And if you want to play it 2,3,... times, you just need to multiply the size by the number of times!

jasonmce, 2025-04-11:
I created a fork and added a one line change to have the player exit gracefully at the end of the song.
https://github.com/jasonmce/PiFmRds-single-play