ChristopheJacquet/PiFmRds #163: Changing PS, RT and TA at run-time - not working

jankonvicny, 2021-03-12:
Hello, 
I try to change RT, PS and TA parameters in runtime, but it`s not working.
After send commands:

sudo ./pi_fm_rds -ctl rds_ctl -freq 87.5 -audio pulses.wav
echo "RT NewText" >rds_ctl
cat >rds_ctl

but nothing happened.
After break transmission and send first command  "sudo ./pi_fm_rds -ctl rds_ctl -freq 87.5 -audio pulses.wav" is now visible RT set to: "NewText"

Please help me with this, thank you

ChristopheJacquet, 2021-03-15:
The behavior you're seeing makes me suspect rds_ctl is a regular file. It needs to be a FIFO. Can you try to remove it and to create a FIFO?

```
rm rds_ctl
mkfifo rds_ctl
```

jankonvicny, 2021-03-16:
My file rds_ctl was created like FIFO file, but I removed and created this file again.

After this I have the same problem.

In further tests, I found that it works in the following procedure:

mkfifio rds_ctl  --- must be created in PiFmRds/src dir
cat >rds_ctl --- must be before start transmit and send command in other terminal
sudo ./pi_fm_rds -ctl rds_ctl -freq 87.5 -audio pulses.wav ---other terminal than cat >rds_ctl
now in terminal with cat>rds_ctl send commands like RT newRT, PS newPS, TA OFF

Is there a way to use this, for example using the "echo" command, I want to use FM transmitter over SSH for automation in other program.

Thank you

ChristopheJacquet, 2024-03-02:
No you can't make it work with `echo`.

I have recently changed the code so that Pi-FM-RDS blocks at startup until the other end of the pipe is connected (i.e. you have started `cat >rds_ctl`). You were not the only one to be confused by the previous behavior, hopefully this is easier to understand now.

closed 2024-03-02 by ChristopheJacquet