ChristopheJacquet/PiFmRds #166: PiFmRds script kills Aux output?

SchattenWolf2008, 2021-03-21:
Hello.
I have a Raspberry Pi 4B 4GB, and some transmitters connectet to the 3.5mm Aux Output.
When i start the Script, i hear a annoying beep sound from the 3.5mm aux output, and all audio on the aux output is killed.
When i stop the script, that annoying sound is still there.
Its only gone when restarting the whole raspberry.

How can i fix this?
- how can i change the device where the script creates the beep sound?
- how can i just get rid of that beep sound?

I need that sound output, bc i am running some devices on it.
I really want to run the Script also, so how can i fix it?
Thanks.

SchattenWolf2008, 2021-03-26:
@ChristopheJacquet can you please help me?
I need the Aux output, and i want to run the script.

DrBlackWolf, 2021-03-27:
You are going to joke. I'have done the same that you, but i don't have this problem...
Don't know how to help you...

SchattenWolf2008, 2021-04-01:
its not a joke.
I tried to link some stuff before with alsamixer or something...
Idk what the problem is.

The script does start / trigger something that does an beep sound and send that to the aux output. ..

DrBlackWolf, 2021-04-01:
I'm going to try again this weekend I will tell you then

DrBlackWolf, 2021-04-08:
So, i try again but i don't have any problem... i don't know why you have got this strange noise. Sorry for you...

andimik, 2021-04-08:
Is the "beep" at 9,5 kHz?

SchattenWolf2008, 2021-04-09:
> Is the "beep" at 9,5 kHz?

yes it is exactly

andimik, 2021-04-09:
https://github.com/ChristopheJacquet/PiFmRds/issues/159

Here you can read how to solve it.

The beep is the stereo pilot tone, which is normally at 19 kHz.

SchattenWolf2008, 2021-04-10:
> #159
> 
> Here you can read how to solve it.
> 
> The beep is the stereo pilot tone, which is normally at 19 kHz.

Beep between 7KHz and 7,2KHz, btw the audio is killed completely, you only hear the beep, nothing else..

SchattenWolf2008, 2021-04-13:
> #159
> 
> Here you can read how to solve it.
> 
> The beep is the stereo pilot tone, which is normally at 19 kHz.

well, changing that didnt change mutch. The beep is on 7,2KHz.
By the way the script completely kills my Aux output, so I dont hear anything anymore, only the beep. And that is still when I stop the script.