ChristopheJacquet/PiFmRds #168: "Official" Android App?

hackthedev, 2021-05-15:
Okay so I made an Android App some time ago so I can control it from my Smartphone. My friends say it is cool so I uploaded it to Github and some App Stores.

My question now is if we might add it to the readme or something when my app submittion is done with approval? Its a prototype but it worked on my machine and Smartphone, but wasn't able to test it in a big scale or something.

As mentioned above, I made a Github Repo called "remotepifm". Maybe someone wants to take a look at it, since the complete project I made is open source now

hex007, 2021-05-16:
@hackthedev Please have a look at these two that I had written in 2016

https://github.com/hex007/pifm_server

https://github.com/hex007/pifm_android_client

hackthedev, 2021-05-16:
The App I made is nothing special I guess. It's my first Android Studio project and also my first java project

hex007, 2021-05-16:
Welcome to the world of Android development.

I referenced my older projets so you could have a look and that's all. Those are written in Python and Java respectively.

hackthedev, 2021-05-16:
I had a lot of problems with the file chooser/picker . It's a real pain but I thing but it seems to work now

ChristopheJacquet, 2024-03-02:
Closing as won't fix given the current status of remotepifm (https://github.com/hackthedev/remotepifm/issues/7). Please reopen if you restart development.

closed 2024-03-02 by ChristopheJacquet