ChristopheJacquet/PiFmRds #170: compalation error on raspi 0w

user421921, 2021-08-05:
after I changed into the directory of /PiFmRds/src i did make clean and then make, but when I executed make I got this error:

gcc -Wall -std=gnu99 -c -g -march=armv6 -O3 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=1 rds.c
In file included from /usr/include/features.h:485,
                 from /usr/include/arm-linux-gnueabi/bits/libc-header-start.h:33,
                 from /usr/include/stdint.h:26,
                 from /usr/lib/gcc/arm-linux-gnueabi/10/include/stdint.h:9,
                 from rds.c:21:
/usr/include/arm-linux-gnueabi/gnu/stubs.h:10:11: fatal error: gnu/stubs-hard.h: No such file or directory
   10 | # include <gnu/stubs-hard.h>
      |           ^~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:43: rds.o] Error 1

sijanec, 2021-10-03:
I have the same problem on Debian Bullseye. You also need package `libsndfile1-dev`, but that's not related to this bug.

Try this:
```
cd src/
sed -e s/=hard/=soft/g < Makefile > Makefile2
make -f Makefile2
```

Remanent, 2024-04-17:
have you tried running it using sudo (command)

idamato, 2025-06-22:
dear, I have successfully tested the compilation having done only a small modification as I am using Raspberry Pi Zero 2W. As the cat /proc/device-tree/model gives "Raspberry Pi Zero 2 W Rev 1.0" the grep doesn't collect the right version. 
I forced the variable RPI_VERSION=4 because no specific differences are present.

closed 2025-11-30 by ChristopheJacquet