ChristopheJacquet/PiFmRds #182: Make doesn't make a nothing

throwitthefuckaway, 2022-10-08:
Was on PiFmRds/src , making everything as normal, and I got this weird error:
"expr:syntax error: unexpected argument '1'"

I don't know if this is a error in my side or what

thiagohrm, 2022-11-29:
are you using 32 or 64 bit OS? i was able to make it using Raspbian OS **32 Bits**

ChristopheJacquet, 2024-03-02:
May you paste the complete output of `make`?

BillyPung, 2024-03-12:
I met the same problem.
My device: Rpi Zero 2W, OS: Raspberry Pi OS (Legacy 32-bit)

My command: make clean
> expr: syntax error: unexpected argument ‘1’
> rm -f *.o *_test

Then I try to: make
> expr: syntax error: unexpected argument ‘1’
> gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other rds_strings.c
> gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other rds.c
> gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other waveforms.c
> gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other rds_wav.c
> gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other fm_mpx.c
> gcc  -o rds_wav rds_wav.o rds.o rds_strings.o waveforms.o fm_mpx.o -lsndfile -lm

Remanent, 2024-04-13:
I get the same error using the same pi zero 2 w, w/ pi 0s legacy 32bit lite (no desktop), and dietpi v8 bookworm.

running this:
`cat /proc/device-tree/model | grep -a -o "Raspberry\sPi\s[0-9]" | grep -o "[0-9]"`
returns no result at the command line.

I believe using the model info obtained from cat /proc/device-tree/model is giving a result the script doesn't like.

might using the info for the arm model cat /proc/cpuinfo and have it pull from model name for the arm version?

`cat /proc/cpuinfo`

```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 902120
Serial          : 00000000a6895766
Model           : Raspberry Pi Zero 2 W Rev 1.0
```

further modifying the line to this:
`cat /proc/cpuinfo | grep "model name" | head -n 1 | awk '{print $4}'`
results this:
`ARMv7`

not sure if this would be usable or not, but i hope at least some of this helps.

Remanent, 2024-04-13:
I do have an original pi1, pi1 model b, pi 2b, pi3b, pi4, pi5, and pi zero 2 w, let me know if you need me to test something on any of these.

Remanent, 2024-04-15:
I used chatgpt to replace the detection portion of makefile w/ a choice set of options to let you select the arm version you want to build for, and it ran successfully on my pi zero 2 w

`CC = gcc
STD_CFLAGS = -Wall -std=gnu99 -c -g

# Prompt user to select architecture
ARCH_PROMPT := "Select the architecture (1 for armv6l, 2 for armv7l, 3 for aarch64): "
ARCH_CHOICE := $(shell read -p $(ARCH_PROMPT) arch && echo $$arch)

# Determine hardware platform and set proper compilation flags based on user choice
ifeq ($(ARCH_CHOICE), 1)
    ARCH_CFLAGS = -march=armv6 -O3 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math
    TARGET = 1
else ifeq ($(ARCH_CHOICE), 2)
    ARCH_CFLAGS = -march=armv7-a -O3 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math
    TARGET = 2
else ifeq ($(ARCH_CHOICE), 3)
    ARCH_CFLAGS = -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math
    TARGET = 3
else
    $(error Invalid architecture choice. Please choose 1 for armv6l, 2 for armv7l, or 3 for aarch64.)
endif

CFLAGS = $(STD_CFLAGS) $(ARCH_CFLAGS) -DRASPI=$(TARGET)

ifneq ($(TARGET), 4)

app: rds.o waveforms.o pi_fm_rds.o rds_strings.o fm_mpx.o control_pipe.o mailbox.o
	$(CC) $(LDFLAGS) -o pi_fm_rds rds.o rds_strings.o waveforms.o mailbox.o pi_fm_rds.o fm_mpx.o control_pipe.o -lsndfile -lm

endif

rds_wav: rds.o rds_strings.o waveforms.o rds_wav.o fm_mpx.o
	$(CC) $(LDFLAGS) -o rds_wav rds_wav.o rds.o rds_strings.o waveforms.o fm_mpx.o -lsndfile -lm

rds_strings.o: rds_strings.c rds_strings.h
	$(CC) $(CFLAGS) rds_strings.c

rds_strings_test: rds_strings.o rds_strings_test.c
	$(CC) -Wall -std=gnu99 -o rds_strings_test rds_strings.o rds_strings_test.c
	./rds_strings_test

rds.o: rds.c waveforms.h rds_strings.o
	$(CC) $(CFLAGS) rds.c

control_pipe.o: control_pipe.c control_pipe.h rds.h
	$(CC) $(CFLAGS) control_pipe.c

waveforms.o: waveforms.c waveforms.h
	$(CC) $(CFLAGS) waveforms.c

mailbox.o: mailbox.c mailbox.h
	$(CC) $(CFLAGS) mailbox.c

pi_fm_rds.o: pi_fm_rds.c control_pipe.h fm_mpx.h rds.h mailbox.h
	$(CC) $(CFLAGS) pi_fm_rds.c

rds_wav.o: rds_wav.c
	$(CC) $(CFLAGS) rds_wav.c

fm_mpx.o: fm_mpx.c fm_mpx.h
	$(CC) $(CFLAGS) fm_mpx.c

clean:
	rm -f *.o *_test`

0xMH09, 2024-04-17:
I have the same issue, i edited the make file and tried all three options, and running 
```
cat /proc/cpuinfo or cat /proc/device-tree/model
``` 
didnt give me the ARM type, im using rasberrypi os 64 bit Lite with rasberry pi zero 2

Remanent, 2024-04-17:
It failed for me on my pi zero 2 w, using raspberry os lite 64bit, unless you need a 64 bit application, try 32bit (the device only has 512mb ram anyways, no real need for 64 bit address space unless you have a program tha tneeds 64bit)

Remanent, 2024-04-18:
It failed for me on my pi zero 2 w, using raspberry os lite 64bit, unless you need a 64 bit application, try 32bit (the device only has 512mb ram anyways, no real need for 64 bit address space unless you have a program that needs a 64bit OS)  Given

0xMH09, 2024-04-18:
it worked with a 32bit OS!

03stevensmi, 2024-05-24:
@ChristopheJacquet hey, Christopher... Can you please look into this?
I'm getting the same error posted above on a raspberry pi 2 using raspberry os legacy 32bit!

aldoniel, 2024-09-01:
Bonjour, Hi.
It seems that there is a syntax error actually in the makefile for raspberry 2W 32 bits. It seems I had the same error as supra.
Previously I opened an issue but closed it because it's a duplicate.
I'm reposting here my issue and my dirty  workaround.

### Error description : 

- followed installation recipe
- `make clean` returns
```
expr: syntax error: unexpected argument ‘1’
rm -f *.o *_test
```
- `make` returns
```
make
expr: syntax error: unexpected argument ‘1’
gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other rds_strings.c
gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other rds.c
gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other waveforms.c
gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other rds_wav.c
gcc -Wall -std=gnu99 -c -g -O3 -DRASPI=other fm_mpx.c
gcc  -o rds_wav rds_wav.o rds.o rds_strings.o waveforms.o fm_mpx.o -lsndfile -lm

```
- then
```
sudo ./pi_fm_rds
sudo: ./pi_fm_rds: command not found

```
### Workaround
- harcoded two variable in Makefile because i'm not proficient with the evil makefile syntax.

```
#UNAME := $(shell uname -m)
UNAME := 'armV7l'

#RPI_VERSION := $(shell cat /proc/device-tree/model | grep -a -o "Raspberry\sPi\s[0-9]" | grep -o "[0-9]")
RPI_VERSION := 2
```
- after make clean and make, the program compiles and runs normally.

Merci beaucoup pour votre programme.

MaffooBristol, 2025-09-10:
I have a Pi Zero 2 W Rev 1.0 and the one line in the Makefile that was causing issues for me was the one where it tries to get the version number, because the regex isn't inclusive enough to accept the word "zero" in there. This can be tweaked as follows, and worked for me:

```diff
diff --git a/src/Makefile b/src/Makefile
index 9e92a11..425f16f 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -5,7 +5,7 @@ STD_CFLAGS = -Wall -std=gnu99 -c -g
 UNAME := $(shell uname -m)

 # Determine Raspberry Pi version (if 2 or greater)
-RPI_VERSION := $(shell cat /proc/device-tree/model | grep -a -o "Raspberry\sPi\s[0-9]" | grep -o "[0-9]")
+RPI_VERSION := $(shell cat /proc/device-tree/model | grep -a -o "Raspberry\sPi\s\(Zero\s[0-9]\|[0-9]\)" | grep -o "[0-9]")

 # Determine the hardware platform and set proper compilation flags
 ifeq ($(UNAME), armv6l)
```