ChristopheJacquet/PiFmRds #188: Hardware related issue with additional FM transmitter

fleshgordo, 2023-07-21:
I was connecting a [15W FM Transmitter](https://www.aliexpress.com/item/1005003658038011.html?spm=a2g0o.order_list.order_list_main.193.34715c5fklMFUY)  to GPIO 7 with a SMA cable and plug in order to boost transmission power. As soon as I connect the 15W transmitter the raspberry power-cycles and won't boot up again. I wonder if the FM transmitter should be separated by an optocoupler or if a diode in the feedline would make sense. I don't have an oscilloscope to measure but I assume that the FM transmitter is feeding something back into GPIO4 which causes an instable behavior on the rpi. Anyone had experience with this?

KubaPro010, 2023-08-29:
A antenna goes to pin 7 (gpio 4), not a transmitter

fleshgordo, 2023-08-29:
Sure, i tried an antenna and it worked well. I'd need to amplify the power output though, hence the PA.

RayRaytheGrape, 2023-09-26:
What are you using to power the amplifier?

KubaPro010, 2023-09-26:
@fleshgordo you can use a power amplifier, but NOT a transmitter, the pi itself is the transmitter

closed 2024-02-24 by ChristopheJacquet