ChristopheJacquet/PiFmRds #189: compile issue

Thebytebeyond, 2023-10-10:
Hello, i am experiencing issues with compiling your code with the make command.

this is the output:

gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 rds.c
rds.c: In function 'set_rds_rt':
rds.c:240:5: warning: 'strncpy' specified bound 64 equals destination size [-Wstringop-truncation]
  240 |     strncpy(rds_params.rt, rt, 64);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
rds.c: In function 'set_rds_ps':
rds.c:247:5: warning: 'strncpy' specified bound 8 equals destination size [-Wstringop-truncation]
  247 |     strncpy(rds_params.ps, ps, 8);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 waveforms.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 pi_fm_rds.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 fm_mpx.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 control_pipe.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 mailbox.c
gcc -o pi_fm_rds rds.o waveforms.o mailbox.o pi_fm_rds.o fm_mpx.o control_pipe.o -lm -lsndfile

This is identical to issue #119 which appears to have been fixed by #117 except for my attempt. I have examined the makefile and there is provison for uname -m to return aarch64 which it odes in my case. i am on a pi4b  would that make a difference? Any help is welcome, cheers.

ChristopheJacquet, 2024-03-02:
I see only two warnings here (which I have fixed recently btw), but no error. So chances are it's just working as intended and you can now run `pi_fm_rds` with `sudo`.

Thebytebeyond, 2024-03-03:
was just warnings. I think main problem was I needed to compile in PiFmRds/src and use the command there

closed 2024-03-03 by Thebytebeyond