ChristopheJacquet/PiFmRds #191: Instant crash on launch on Pi 4

creeper6530, 2024-01-07:
Immedeatelly after I start transmitting, my whole Pi freezes, doesn't transmit and simply stalls until the watchdog forces a reboot. I am on latest commit and with up-to-date Raspbian running on a Pi 4 (using only `apt update && apt full-upgrade --auto-remove`).

Compile output:
`$ make clean && make -j`

```make
rm -f *.o
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 rds.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 waveforms.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 pi_fm_rds.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 fm_mpx.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 control_pipe.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 mailbox.c
mailbox.c: In function ‘mapmem’:
mailbox.c:65:35: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   65 |         printf("mmap error %d\n", (int)mem);
      |                                   ^
pi_fm_rds.c: In function ‘mem_phys_to_virt’:
pi_fm_rds.c:295:45: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  295 |     return phys - (uint32_t)mbox.bus_addr + (uint32_t)mbox.virt_addr;
      |                                             ^
pi_fm_rds.c: In function ‘tx’:
pi_fm_rds.c:340:49: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘long unsigned int’ [-Wformat=]
  340 |     printf("Allocating physical memory: size = %d     ", NUM_PAGES * 4096);
      |                                                ~^
      |                                                 |
      |                                                 int
      |                                                %ld
pi_fm_rds.c:447:24: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  447 |     uint32_t last_cb = (uint32_t)ctl->cb;
      |                        ^
pi_fm_rds.c:509:38: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  509 |         int last_sample = (last_cb - (uint32_t)mbox.virt_addr) / (sizeof(dma_cb_t) * 2);
      |                                      ^
pi_fm_rds.c:510:37: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  510 |         int this_sample = (cur_cb - (uint32_t)mbox.virt_addr) / (sizeof(dma_cb_t) * 2);
      |                                     ^
pi_fm_rds.c:540:19: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  540 |         last_cb = (uint32_t)mbox.virt_addr + last_sample * sizeof(dma_cb_t) * 2;
      |                   ^
rds.c: In function ‘set_rds_rt’:
rds.c:240:5: warning: ‘strncpy’ specified bound 64 equals destination size [-Wstringop-truncation]
  240 |     strncpy(rds_params.rt, rt, 64);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
rds.c: In function ‘set_rds_ps’:
rds.c:247:5: warning: ‘strncpy’ specified bound 8 equals destination size [-Wstringop-truncation]
  247 |     strncpy(rds_params.ps, ps, 8);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -o pi_fm_rds rds.o waveforms.o mailbox.o pi_fm_rds.o fm_mpx.o control_pipe.o -lm -lsndfile
```

Output of `sudo rpi-eeprom-update`:
```
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Tue 26 Apr 10:24:28 UTC 2022 (1650968668)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
```
I never upgraded my firmware, should I? Will it fix/change something? Using markonej/fm_transmitter works without problem.

ChristopheJacquet, 2024-02-17:
I don't have a RPi 4 to test myself, so I'm not sure.

What's the output of the `arch` command? I'm wondering if maybe it could be an incompatibility with 64-bit systems.

creeper6530, 2024-02-17:
I'm on `aarch64`, so yes, a 64-bit system

ChristopheJacquet, 2024-03-02:
Can you try with the latest version from the repository? I have fixed all the warnings above, some of which could lead to bugs on 64 bits potentially. The bugs did not manifest on my RPi 3, but maybe they did on RPi 4...

creeper6530, 2024-03-02:
It compiles without any warnings (again `make clean && make -j`):

```make
rm -f *.o *_test
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 rds_strings.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 waveforms.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 pi_fm_rds.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 fm_mpx.cgcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 control_pipe.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 mailbox.c
gcc -Wall -std=gnu99 -c -g -march=armv8-a -O2 -pipe -fstack-protector-strong -fno-plt -ffast-math -DRASPI=4 rds.c
gcc  -o pi_fm_rds rds.o rds_strings.o waveforms.o mailbox.o pi_fm_rds.o fm_mpx.o control_pipe.o -lsndfile -lm
```

I'll test it as soon as I get home, this was done from SSH.

creeper6530, 2024-03-02:
After running the following command:

`sudo ./pi_fm_rds -audio /mnt/nfs/(...)/music/sample.wav -freq 91.5 | tee /home/(...)/PiFmRds/src/debug.out`

It immedeately crashed again (SSH disconnected, Pi rebooted probably due to watchdog triggering). `debug.out` didn't even manage to get created (or rather saved to SD card) most of the time, or only got saved as an empty file.

Radio turned silent for few seconds, then returned to static.

When I tried it another time, it started playing for half a second, then started looping the same tenth of second (the ta-ta-ta-ta-ta-ta-ta sound) before finally turning back to static as the watchdog kicked in. The SSH session froze up and timed out as well.

I'm sure the usage of NFS didn't cause it, as it runs over 1Gb LAN over to a quite fast NAS that only I use. It might be even faster with than local SD card. The fact that it managed to transmit a small chunk of it once (see previous paragraph) only solidifies this theory.

TL;DR: Nothing changed and I didn't manage to catch any output, sadly. markonej/fm_transmitter still works on up-to-date main branch.