ChristopheJacquet/PiFmRds #193: lack fpu

mrdoitall2712, 2024-02-22:


ChristopheJacquet, 2024-02-24:
Can you elaborate on that?

mrdoitall2712, 2024-02-27:
when i make file it says lacks fpu and if i get it working on raberry os 32 bit legacy it sounds well for five seconds and then static takes over

ChristopheJacquet, 2024-03-02:
What Raspberry Pi model are you using? What OS? You can get all these by running:

```
cat /etc/os-release
uname -a
cat /sys/firmware/devicetree/base/model
```
Could you also paste the complete `make` output when it fails with "lacks fpu"?

he1313, 2024-07-01:
hi,i have the same problem . when i make it .it is show  

[image]

[image]

KubaPro010, 2024-09-02:
This is a architecture related problem, ARM CPUs sometimes cheap out and leave the Floating Point Unit out, so get a another pi

Sorzus-michael, 2024-10-04:
Did you manage to solve it ? i'm having the same on my 4b