ChristopheJacquet/PiFmRds #200: 19k Signal and Increasing RDS Power

danielmoore123, 2024-05-20:
I noticed that the program would always play mono even with a stereo source. 
I tried adding -c 2 but mono. 

When looking at it in SDR Sharp, spotted there is no 19k but oddly is an "M shaped" 57k, presuming this is the RDS?
Is there anyway of modifying this or is this a result of a "square wave"?

In addition to this, I note some recievers can pick up the RDS signal on Pi3 but not Pi2 or Pi1, I was wondering if there was a way to tweak the RDS power? On a usual encoder, this can be increased and decreased often with a dial, appreciate this isn't the "real deal" but good for exploring the spectrum.

danielmoore123, 2024-05-20:
I realised what I did wrong!

The 19k signal only comes on in stereo... As soon as it's present, the recievers not picking up the RDS suddenly respond! 

Interesting to know some have RDS on mono signals whilst others don't!

closed 2024-05-20 by danielmoore123

ryangontv, 2024-05-21:
Hi Daniel, this "M shaped" signal at 57kHz is indeed RDS.