ChristopheJacquet/PiFmRds #201: Add gui

Stormtrooper0744, 2024-06-24:


satana256, 2024-06-27:
I have a device using this utility. and there is no display, I use an ssh connection. and it seems to me that this would be a pretty wild idea with a Gui shell., but the very essence is good. That's not the point. if the author of the utility has the time and desire, I have no doubt he will publish the result. sorry for the crooked translation from the Ru language

douxxu, 2024-09-04:
Hey, actually i created a minimal webgui for this utility for my personal uses, so maybe it can help you, here is the github link:
https://github.com/douxxu/piwave-webgui

I don't wan't to be disrespectful promoting my content, so if the author didn't like it, feel free to delete it

furipaf, 2025-02-05:
Do you need Web Application to control all this stuff? check this, I made it and trying in my spare time to add ore features:

https://github.com/furipaf/PiFmRds-WebApp

ChristopheJacquet, 2025-11-30:
I'm closing this as adding a GUI is out of the scope of this experimental project. Thanks @furipaf for providing a link to your nice UI.